mcu-tools/mcuboot #2151: find_last_sector_idx never returning

lefebvresam, 2024-12-17:
After calling `boot_read_image_header(&state, 0, &hdr, &status);` there is a call `find_swap_count(state, swap_size);` and `find_last_sector_idx(state, copy_size);` with copy_size = 76040.

This code part is never converging and crashes (line 481 of swap_scratch.c):

```
    while (1) {
        if ((primary_slot_size < copy_size) ||
            (primary_slot_size < secondary_slot_size)) {
           primary_slot_size += boot_img_sector_size(state,
                                                     BOOT_PRIMARY_SLOT,
                                                     last_sector_idx);
        }
        if ((secondary_slot_size < copy_size) ||
            (secondary_slot_size < primary_slot_size)) {
           secondary_slot_size += boot_img_sector_size(state,
                                                       BOOT_SECONDARY_SLOT,
                                                       last_sector_idx);
        }
        if (primary_slot_size >= copy_size &&
                secondary_slot_size >= copy_size &&
                primary_slot_size == secondary_slot_size) {
            break;
        }
        last_sector_idx++;
    }
```

It should look more logical that this must be:

```
        if ((primary_slot_size >= copy_size &&
                secondary_slot_size >= copy_size) ||
                primary_slot_size == secondary_slot_size) {
            break;
        }
```

And the piece of code should have a check to assert before crashing.

nordicjm, 2024-12-18:
This code works fine on zephyr and mynewt so you have a problem with your deprecated and unsupported mbedos code which is outside the scope of MCUboot

lefebvresam, 2024-12-18:
swap_scratch.c is part of bootutil and has nothing to do with mbed. 

[@multiplemonomials](https://github.com/multiplemonomials) asked to restart the mbed-ce support in [pull/2152](https://github.com/mcu-tools/mcuboot/pull/2152) becasue we have our community port of mbed-os.

nordicjm, 2024-12-18:
> swap_scratch.c is part of bootutil and has nothing to do with mbed.

The code above adds sector sizes to a count value and that code is definitely coming from mbed, mcuboot has to query it somehow so your mbed code has an issue there (i.e. maybe it is returning 0 if you enter an infinite loop)

> [@multiplemonomials](https://github.com/multiplemonomials) asked to restart the mbed-ce support in [pull/2152](https://github.com/mcu-tools/mcuboot/pull/2152) becasue we have our community port of mbed-os.

Interesting, if they're willing to maintain support for it and look into the issue then that's fine

lefebvresam, 2024-12-18:
I do not see any mbed references in this code:

```
 * SPDX-License-Identifier: Apache-2.0
 *
 * Copyright (c) 2019 JUUL Labs
```

nordicjm, 2025-01-24:
@multiplemonomials can you check if this can be closed?

multiplemonomials, 2025-01-28:
@lefebvresam Do you have a project that reproduces this issue that is based on current master branch of mcuboot (i.e. one that uses the CMake build system)? If so I can give it a shot on my STM32U5 board and see if I can reproduce the error

lefebvresam, 2025-01-28:
@multiplemonomials  This was in the context of [version extraction](https://github.com/multiplemonomials/mcuboot/pull/1). In the beginning there was no initialization of state and boot status which led to unlogical decisions in the calling tree. After doing such initalization as proposed in the patch, this problem was not occuring anymore. However, I never went into the deep details of checking primary/secundary slot size in this piece of code and I still find it a strange construction which can potentially lead to hanging software. I should recommend to revise this code and put some additional checks or assertions to avoid hanging behavior.

lefebvresam, 2025-01-28:
If you want to reproduce you can Cherry pick from [here](https://github.com/multiplemonomials/mcuboot/pull/1/commits/8b92593b1f9e3dffadbb1bdb8906ad7a1423335b) I think. Taking into accoutn the remarks above.

closed 2026-01-04 by github-actions[bot]