mcu-tools/mcuboot #2172: swap_move: Message about non optimal sector distribution seems wrong

mimok, 2025-01-11:
This is not a critical issue but i believe there is a problem with the test which checks whether the sector distribution is optimal or not.
This test can be found here: https://github.com/mcu-tools/mcuboot/blob/59e9f03bb3a78c78bdf100d66e9d088a344e7c33/boot/bootutil/src/swap_move.c#L286

On my STM32F429 MCU, I have seven 128KB sectors for slot 0, six 128KB sectors for slot 1 and mcuboot complains about non optimal distribution of sectors while  everything seems to work.

If i put five sectors in slot1, mcuboot does not complain anymore but the maximum size of an update is reduced by 128KB.

Why a 7/6 distribution would be less optimal than a 7/5 distribution ?

nordicjm, 2025-01-13:
The check is correct, in swap using move mode you need to move all sectors up by one in the primary slot, and you need to store the swap status for each sector, for the first part you just take away one sector, for the second part you take away at least one sector, assuming you have a write block size of e.g. 32 bytes and either `CONFIG_BOOT_MAX_IMG_SECTORS_AUTO` is set or `CONFIG_BOOT_MAX_IMG_SECTORS` is set to 7, there are 3 states so that is 7 * 32 * 3 = 672 bytes for just the swap status part alone, it will be larger than that because of the other fields (see https://docs.mcuboot.com/design.html#image-trailer for details) but it will be less than a 128KiB sector's worth, therefore 2 sector overhead when using swap using move mode. For such large sector sizes, swap using move is not ideal, if you can forego application reverts then upgrade only mode is better, or if not, direct-xip with revert would allow larger application sizes without swapping but would need updating of the correct image slot

mimok, 2025-01-13:
Thank you @nordicjm for your answer. However, slot 1 also contains a trailer which takes one sector. In consequence, there is an overhead of two sectors for slot 0 and one sector flot slot 1  and the difference is only one sector. Well, it's my understanding of the source code: in `boot_move_sector_up()` both trailers of slot 0 and slot 1 are erased and the content of slot 0 is moved by one sector.

closed 2025-08-06 by github-actions[bot]