mcu-tools/mcuboot #2179: NuttX mcuboot_config.h missing RAM_START and RAM_SIZE for RAM_LOAD?

TimJTi, 2025-01-15:
I am wanting to use MCUBOOT with a NuttX project, using RAM load. I believe there is an error in [mcuboot_config.h](https://github.com/mcu-tools/mcuboot/blob/main/boot/nuttx/include/mcuboot_config/mcuboot_config.h) for NuttX.

It is currently:

```
#ifdef CONFIG_MCUBOOT_RAM_LOAD
#  define MCUBOOT_RAM_LOAD
#endif
```
but I think it should be more like the Zephyr variant of this file:

```
#ifdef CONFIG_BOOT_RAM_LOAD
#  define MCUBOOT_RAM_LOAD 1
#  define IMAGE_EXECUTABLE_RAM_START CONFIG_BOOT_IMAGE_EXECUTABLE_RAM_START
#  define IMAGE_EXECUTABLE_RAM_SIZE CONFIG_BOOT_IMAGE_EXECUTABLE_RAM_SIZE
#endif
```

I have created appropriate Kconfig entries relevant to my board ("platform") for `BOOT_IMAGE_EXECUTABLE_RAM_START ` and `BOOT_IMAGE_EXECUTABLE_RAM_SIZE`

I am still early on with my journey using mcuboot so if I'm incorrect please advise - but making this change does, at least, allow my early mcuboot/NuttX test code to compile without the #error macro for these config entries.

I can submit a PR if I am right, and if wanted.

de-nordic, 2025-01-23:
Have you experimented with the setting in your setup?

TimJTi, 2025-01-23:
> Have you experimented with the setting in your setup?

@de-nordic 

I am still trying to get MCUboot to work for my NuttX-based custom board and app, but it just won't compile without those two CONFIG settings if I choose that MCUboot mode. No NuttX examples/boards have references to these for me to compare to and I imagine it is just an oversight/omission if no one has tried this MCUboot variation.

That said, I am not sure I  actually need it, now I know more from my experimentations. I was thinking it was necessary to use this mode to copy the app code stored in flash to DDRAM to execute but I think I was mistaken and it is is more to do with verifying code than  copying/running code.

Uboot works (but the configuration of that I did years ago and can't remember the details) and so far I have MUCboot recognise the slot image I have created as valid, and it tries to load it but then my board crashes. I think this is probably due to DDRAM region conflicts between the AT91 bootstrap, MCUBoot, and my NuttX build/app. I will be back on this tomorrow.

So, as far as this report is concerned, I still think it is an omission but not a show stopper for me.