mcu-tools/mcuboot #2183: boot_copy_region uses boot_encrypt before boot_enc_init if a swap is continued

Olstyle, 2025-01-21:
`swap_run` will reach down to  `boot_copy_region` which always calls `boot_encrypt ` for encrypted images, but the AES context is only initialized via `boot_enc_load` if a swap was NOT ongoing.
This effectively breaks swap continuation for encoded images if the used crypto library relies on initialization.
Which is the case for mbedtls_aes_init and will pretty much always be the case for HW based implementations.

The following screenshots show my HW implementation failing, but according to my review, mbedtls will fail at the same point!

[image]
[image]

de-nordic, 2025-01-21:
What about loop doing the `boot_enc_init`  here https://github.com/mcu-tools/mcuboot/blob/06747985eee7c27d891f2529c58e18f4a84ea6a3/boot/bootutil/src/loader.c#L1625-L1635

Olstyle, 2025-01-21:
> What about loop doing the `boot_enc_init` here
> 
> [mcuboot/boot/bootutil/src/loader.c](https://github.com/mcu-tools/mcuboot/blob/06747985eee7c27d891f2529c58e18f4a84ea6a3/boot/bootutil/src/loader.c#L1625-L1635)
> 
> Lines 1625 to 1635 in [0674798](/mcu-tools/mcuboot/commit/06747985eee7c27d891f2529c58e18f4a84ea6a3)
> 
>          if (bs->enckey[slot][i] != 0xff) { 
>              break; 
>          } 
>      } 
>   
>      boot_enc_init(BOOT_CURR_ENC(state), slot); 
>   
>      if (i != BOOT_ENC_KEY_SIZE) { 
>          boot_enc_set_key(BOOT_CURR_ENC(state), slot, bs); 
>      } 
>  }

Your are right, this seems to be the fixed in main, but only recently 
https://github.com/mcu-tools/mcuboot/commit/7e3a1cecd73d30ee1e63e111344f03dd7a1b27ec
I am working on the latest release 2.1.0, which still had that issue

de-nordic, 2025-01-23:
> 
> Your are right, this seems to be the fixed in main, but only recently [7e3a1ce](https://github.com/mcu-tools/mcuboot/commit/7e3a1cecd73d30ee1e63e111344f03dd7a1b27ec) I am working on the latest release 2.1.0, which still had that issue. So actually you already fixed it but didn't make a bugfix release out of it.

True. My bad.

de-nordic, 2025-01-23:
Can we close the issue?

Olstyle, 2025-01-23:
Swap continuation with the favored crypto library (since tinycrypt is discontinued) is broken in the release version.
To me this sounds like a patch release would be reasonable. 
At least that's what I would do in my project.
From the pure POV of main this issue it can be considered solved.

closed 2025-07-23 by de-nordic