mcu-tools/mcuboot #2212: Using MCUBOOT_DATA_SHARING with multiple images prevents jumping into application

pbesson-invn, 2025-02-20:
Hello,

I want to use MCUBOOT_DATA_SHARING feature with nrf5340 which uses multiple images (NET+APP).

But doing so prevents booting to application because `boot_add_shared_data()` in `boot/bootutil/src/loader.c` is called twice from `context_boot_go()` when looping images. Indeed, with nrf5340, MCUBOOT_IMAGE_NUMBER is set to 2.

Second time `boot_add_shared_data()` is called, it returns `SHARED_MEMORY_OVERWRITE` which causes `context_boot_go()` to return an error and abort the boot process.

Anyway to workaround that ?

Thank you.

mcuboot version: https://github.com/mcu-tools/mcuboot/commit/daf2946a0f07

nordicjm, 2025-02-24:
Doesn't really make sense because `boot_save_shared_data()` has this:
```
    if (saved_bootinfo) {
        /* Boot info has already been saved, nothing to do */
        return 0;
    }
...
    if (!rc) {
        saved_bootinfo = true;
    }

    return rc;
```
So if the first image update was OK then the second would be skipped

pbesson-invn, 2025-02-24:
Code you are referring to is not present in mcbuoot/nrf version I am using. It looks like you have added it recently: https://github.com/mcu-tools/mcuboot/commit/bcffc62c66666cadc181dd3b02195ec569814116

So I guess this issue can be closed.

Sorry for the trouble.

closed 2025-02-24 by pbesson-invn

nordicjm, 2025-02-24:
Use `main` of ncs and it should be fixed there