mcu-tools/mcuboot #2219: MCUBoot doesn't jump to primary slot if secondary slot is unavailable

td-frosik, 2025-02-26:
I have the secondary slot placed in the external flash. I would like the MCUBoot to jump to the application in the primary slot even if the external flash is damaged or unavailable. 

Could you add configuration option to ignore error from opening the secondary slot flash area?

nordicjm, 2025-02-26:
If the secondary device is not available then the device/hardware is faulty, the abort of booting is intentional

td-frosik, 2025-02-26:
It would be better to have an option to start the application in the primary slot which is valid and be able to send the information about external flash error to cloud or wherever it's needed.

nordicjm, 2025-02-26:
> It would be better to have an option to start the application in the primary slot which is valid and be able to send the information about external flash error to cloud or wherever it's needed.

You could probably do this in MCUboot already with a hook and returning dummy empty data if the device is not present

td-frosik, 2025-02-26:
Could you show me some example that explains how it can be done?

closed 2025-05-14 by nordicjm