mcu-tools/mcuboot #2220: MCUBoot read active slot in serial recovery mode

td-frosik, 2025-02-26:
I have a board with nRF9160 and nRF52840. I can update the firmware of both application and MCUBoot for nRF52840 over the SMP while the nRF52840 is in serial recovery mode.

How nRF9160 can read the nRF52840's active MCUBoot slot and its image version? This is needed for the user to know which build (S0/S1) should be used for the next MCUBoot firmware update.

nordicjm, 2025-02-27:
If the device is in serial recovery mode then it should be single slot, not direct XIP mode, and you would just upload the signed file

td-frosik, 2025-02-27:
Can the upgradable MCUBoot be in single slot instead of XIP mode? Wouldn't it mean that it overwrites itself?

nordicjm, 2025-02-27:
So by active slot you mean the active slot of the currently running MCUboot instance, not the application. In that case no, and there is no way to get the slot, you would need to add something like an SMP extension command to get the active slot

nvlsianpu, 2025-03-11:
@td-frosik What you are asking here is Nordic Semiconductor fork feature.
If you need that feature you rather should ask on Nordic's DevZone.

td-frosik, 2025-03-12:
You're right, I didn't notice that's a fork.

Related Nordic's DevZone post:
https://devzone.nordicsemi.com/f/nordic-q-a/119194/mcuboot-active-slot-and-version-over-smp

closed 2025-03-12 by td-frosik