mcu-tools/mcuboot #2223: mcuboot espressif port enable Flash Encryption, use zephyr rtos, ble dfu error.

zhang-wenchao, 2025-02-27:
[00:07:44.435,000] <err> mcumgr_img_grp: Faled boot_set_next with code 3, for slot 1, with active slot 0 and confirm 0

nordicjm, 2025-03-14:
@almir-okato

almir-okato, 2025-03-14:
Hi @zhang-wenchao, could you please share more details of which commands you have used to build the Espressif Port, the content of the used `.conf` file and the boot console output?

Also, `CONFIG_MCUBOOT_BOOTLOADER_MODE_DIRECT_XIP_WITH_REVERT` configuration is not supported by MCUboot Espressif Port, so all the execution is not supposed to properly work.

zhang-wenchao, 2025-03-15:
Ok, I'll create a minimal example as soon as possible.

zhang-wenchao, 2025-03-15:
This issue is temporarily closed and will be reopened when I can reproduce it with a minimal example.

zhang-wenchao, 2025-03-15:
> Hi [@zhang-wenchao](https://github.com/zhang-wenchao), could you please share more details of which commands you have used to build the Espressif Port, the content of the used `.conf` file and the [...]

And what operations are supported?

closed 2025-03-15 by zhang-wenchao

zhang-wenchao, 2025-09-10:
CONFIG_MCUBOOT_BOOTLOADER_MODE_DIRECT_XIP_WITH_REVERT 

> Hi [@zhang-wenchao](https://github.com/zhang-wenchao), could you please share more details of which commands you have used to build the Espressif Port, the content of the used `.conf` file and the [...]

Is CONFIG_MCUBOOT_BOOTLOADER_MODE_DIRECT_XIP_WITH_REVERT supported now? Is there any description or code for the available options?