mcu-tools/mcuboot #2224: Measured data are not shared with mainapp when single slot application is configured

MBorto, 2025-02-28:
I was working with two image slots and retention system enabled for shareing bootloader version and hash of slot0 with the mainapp and all worked well.

Now I've enabled the single slot application and measured boot information are not shared by the bootloader.

Investigating the code of the bootloader seems that the loader.c file adds the shared data with:

(https://github.com/mcu-tools/mcuboot/blob/7330df7c783b0da75190459860257fc9f758fd50/boot/bootutil/src/loader.c#L2525)

Instead the file single_loader.c seems not calling this method.

MBorto, 2025-03-03:
I've solved the problem modifing single_loader.c as follows (sorry but i cannot load *.c files directly)

[single_loader.txt](https://github.com/user-attachments/files/19050646/single_loader.txt)

If it's ok you can directly add into MCUBoot project.

nordicjm, 2025-03-04:
Not valid, was added 3 months ago https://github.com/mcu-tools/mcuboot/blob/main/boot/zephyr/single_loader.c#L151

closed 2025-03-04 by nordicjm