mcu-tools/mcuboot #2258: Error: 'mbedtls_asn1_buf' has no member named 'private_len'

joelguittet, 2025-04-09:
Hello,

Building mcuboot (custom project context - no zephyr or other) with MCUBOOT_SIGN_EC256 I get the following error:

```
In file included from /path/to/mcuboot/boot/bootutil/src/image_ecdsa.c:36:
/path/to/mcuboot/boot/bootutil/include/bootutil/crypto/ecdsa.h: In function 'bootutil_import_key':
/path/to/mcuboot/boot/bootutil/include/bootutil/crypto/ecdsa.h:105:12: error: 'mbedtls_asn1_buf' has no member named 'private_len'
  105 |     if (alg.MBEDTLS_CONTEXT_MEMBER(len) != sizeof(ec_pubkey_oid) - 1 ||
      |            ^
/path/to/mcuboot/boot/bootutil/include/bootutil/crypto/ecdsa.h:106:19: error: 'mbedtls_asn1_buf' has no member named 'private_p'
  106 |         memcmp(alg.MBEDTLS_CONTEXT_MEMBER(p), ec_pubkey_oid, sizeof(ec_pubkey_oid) - 1)) {
      |                   ^
/path/to/mcuboot/boot/bootutil/include/bootutil/crypto/ecdsa.h:110:14: error: 'mbedtls_asn1_buf' has no member named 'private_len'
  110 |     if (param.MBEDTLS_CONTEXT_MEMBER(len) != sizeof(ec_secp256r1_oid) - 1 ||
      |              ^
/path/to/mcuboot/boot/bootutil/include/bootutil/crypto/ecdsa.h:111:21: error: 'mbedtls_asn1_buf' has no member named 'private_p'
  111 |         memcmp(param.MBEDTLS_CONTEXT_MEMBER(p), ec_secp256r1_oid, sizeof(ec_secp256r1_oid) - 1)) {
```

It's internals of mcuboot and so I don't understand why I get this error. No problem when using RSA or ED25519.

Joel

taltenbach, 2025-04-09:
Hi Joel! Which version of MCUboot and mbedTLS are you using? It's look like you might be missing this commit 1d79ef35ee560e24da981d13c9414de9947b07a2 (from the MCUboot's main branch).

joelguittet, 2025-04-09:
Thanks for the quick answer. Need to test, but yes you're right.
I have mbedtls 3.6 and mcuboot v2.1. This is the latest release of mcuboot, but the commit has been integrated after, unfortunately.
Just checked the next milestone seems v2.2 can come soon ?

taltenbach, 2025-04-09:
I don't know the planned release date, but since the 2.2.0-rc0 seems to be on the way, I'd expect to see the 2.2.0 quite soon, indeed :)

joelguittet, 2025-04-12:
Fix updating to the latest commit of the master branch.
Closing.

closed 2025-04-12 by joelguittet