mcu-tools/mcuboot #2292: MCUBOOT swap process

LarryHuLiTao, 2025-05-06:
I'm using Renesas RA8D1 for OTA and implementing it with the MCUBOOT architecture. Currently, I'm using the SWAP mode. I want to know the process of MCUBOOT swapping firmware. If there is a sudden power outage during the swapping process, will the MCU become a brick? Does MCUBOOT use a recovery mechanism internally? Are there any more detailed documents about SWAP and porting MCUBOOT that can be referred to?

taltenbach, 2025-05-06:
Please refer to https://github.com/mcu-tools/mcuboot/blob/main/docs/design.md.

To answer your questions:
> If there is a sudden power outage during the swapping process, will the MCU become a brick?

No, MCUboot will resume the swapping process at next boot.

> Does MCUBOOT use a recovery mechanism internally?

Yes, it uses areas called "trailers" at the end of the slots to determine where the process has to be resumed in case it is interrupted due to a power outage.

> Are there any more detailed documents about SWAP and porting MCUBOOT that can be referred to?

All the documentation is in the `docs` folder: https://github.com/mcu-tools/mcuboot/blob/main/docs

LarryHuLiTao, 2025-05-06:
Thank you very much for your answer. After reading the document introduction, I'm not very clear about the rollback mechanism after power failure and re-power-on. Is there another dedicated variable record that has been swapped to that area and needs to be synchronized to the flash in real time because the variable cannot be saved after power failure? Could you please tell me which specific variable it is?

taltenbach, 2025-05-07:
First, note that if a power failure occurs during the upgrade process, no rollback will be performed. MCUboot will take care of resuming and completing the interrupted upgrade process. The rollback feature is not intended to recover from an interrupted upgrade but to be able to rollback to the previous firmware image in case the new one is not working as expected (e.g. because it crashes at boot).

So, regarding the need of storing some metadata to flash, you're right. We can see an upgrade or rollback process as a sequence of steps, each step being a point from which the process can be resumed. After completing each step, MCUboot writes a flag to flash memory to be able to determine where to resume the process in case the upgrade or rollback is interrupted by a power failure. All MCUboot metadata is stored in what MCUboot calls "image trailers" or simply "trailers", that are basically portions of flash memory located at the very end of the primary and secondary slots. The trailer areas are dedicated to MCUboot and cannot be used to store the firmware image.

This is explained in details in the "Image Trailer" section of the design document and the sections that follow.
See: https://github.com/mcu-tools/mcuboot/blob/main/docs/design.md#image-trailer