mcu-tools/mcuboot #2295: MCUboot incompatible with image encryption on ESP32S3

hobbes-400, 2025-05-07:
Performing updates when flash encryption is enabled on an ESP32s3 results in the image trailer (specifically `image_ok`) getting into a bad state. MCUboot attempts to perform a decrypted flash read on an erased trailer (raw `0xFF`), causing it to think that the erased trailer has valid data in it (decrypted `0xFF`). Some details:

In MCUBoot's bootutil_public.c, the boot_read_flag function attempts to read a trailer flag and determine its state:

```
    static int
    boot_read_flag(const struct flash_area *fap, uint8_t *flag, uint32_t off)
    {
        int rc;
        rc = flash_area_read(fap, off, flag, sizeof *flag);
        if (rc < 0) {
            return BOOT_EFLASH;
        }
        if (bootutil_buffer_is_erased(fap, flag, sizeof *flag)) {
            *flag = BOOT_FLAG_UNSET;
        } else {
            *flag = boot_flag_decode(*flag);
        }
        return 0;
    }
```

This function relies on `bootutil_buffer_is_erased()` to check whether the trailer flag buffer is erased. If the buffer is erased, it assumes the flag is unset.

The problem arises because `flash_area_read()` performs a decrypted flash read. When the flash is freshly erased, which happens when image swapping occurs, its raw contents are `0xFF`. However, on an encrypted board, this erased flash (`0xFF`) is attempted to be decrypted. Decrypting `0xFF` does not necessarily yield `0xFF` — it produces arbitrary data. This causes MCUBoot to incorrectly interpret the trailer as containing valid data, rather than being erased, when flash encryption is enabled. Of course, the "valid data" is decrypted garbage, getting the trailer into a corrupted state.

Despite this fact, the decrypted flash read is required in some cases. For instance, `imgtool` creates an "erased" trailer filled with `0xFF` when OTA assets are first generated. When this asset is flashed over-the-air, the `0xFF` values are automatically encrypted. Upon reading them, MCUBoot correctly decrypts the contents back to `0xFF` and correctly recognizes them as erased, setting the flag to `BOOT_FLAG_UNSET` for newly flashed update assets (that have not been swapped / moved yet).

In summary, it appears `boot_read_flag` only supports decrypted flash flows OR encrypted flash flows that have been pre populated with valid data. It does not support flows that have encrypted flash that have been erased. More concretely, when flash encryption is enabled, `bootutil_buffer_is_erased` is more accurately `bootutil_buffer_is_unset` as the read APIs do not take into account the raw flash contents.

We were able to verify this theory with the following patch:

```
     static int
     boot_read_flag(const struct flash_area *fap, uint8_t *flag, uint32_t off)
     {
         int rc;
         rc = flash_area_read(fap, off, flag, sizeof *flag);
         if (rc < 0) {
             return BOOT_EFLASH;
         }
    -    if (bootutil_buffer_is_erased(fap, flag, sizeof *flag)) {
    -        BOOT_LOG_INF("buffer erased");
    +    uint8_t raw_flag;
    +    rc = flash_area_read_raw(fap, off, &raw_flag, sizeof *flag); // perform raw / non-decrypted flash read
    +    if (rc < 0) {
    +        return BOOT_EFLASH;
    +    }
    +
    +    bool raw_val_erased = bootutil_buffer_is_erased(fap, &raw_flag, sizeof *flag);
    +    bool decrypt_val_erased = bootutil_buffer_is_erased(fap, flag, sizeof *flag);
    +    if (raw_val_erased || decrypt_val_erased) {
    +        BOOT_LOG_INF("buffer erased. raw_val_erased:%d val_erased:%d", raw_val_erased, val_erased);

              *flag = BOOT_FLAG_UNSET;
```

Potential solution:
For flash encryption flows, manually set `image_ok` and other relevant flags to UNSET after they have been erased / copied over. A flow similar to this is already used for confirmed image updates. In this case, `image_ok` is set to 1 after the copy is done.

Of note:
If we attempt to OTA a confirmed image, the OTA works as expected as `image_ok` is overwritten during the swap process, leaving it in a known, good state.

This flow occurs both when the OTA asset is updated via BLE (smpmgr) and when hardflashed using esptool. Information for the latter is provided below.

Commits used (patches for the commits linked below):
MCUBoot: commit 346f7374ff4467e40b5594658f8ac67a5e9813c9
Zephyr: commit 7823374e872145b5bd018bfe447839eb36042611 (tag: v4.1.0)
ESP-IDF: commit d7b0a45ddbddbac53afb4fc28168f9f9259dbb79 (HEAD, tag: v5.1.4)

[... 108 lines cut ...]
[esp32s3] [INF] br_image_off = 0x20000
[esp32s3] [INF] ih_hdr_size = 0x20
[esp32s3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32s3] [INF] DRAM segment: start=0x316c0, size=0x261c, vaddr=0x3fc95650
[esp32s3] [INF] IRAM segment: start=0x20080, size=0x11640, vaddr=0x40374000
[esp32s3] [INF] start=0x403803c0
I (9282) boot: IROM segment: paddr=00040000h, vaddr=42000000h, size=3548Eh (218254) map
I (9283) boot: DROM segment: paddr=00080000h, vaddr=3c040000h, size=0A1A0h ( 41376) map
I (9298) boot: libc heap size 240 kB.
I (9298) spi_flash: detected chip: generic
I (9299) spi_flash: flash io: dio

*** Booting Zephyr OS build v4.1.0-4-gecd4e9c68a7b ***
[00:00:09.680,000] <inf> littlefs: LittleFS version 2.10, disk version 2.1
[00:00:09.681,000] <inf> littlefs: FS at flash-controller@60002000:0x3b0000 is 48 0x1000-byte blocks with 512 cycle
[00:00:09.681,000] <inf> littlefs: partition sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:09.682,000] <inf> esp32_bt_adapter: BT controller compile version [fd62b31]
[00:00:09.720,000] <inf> bt_hci_core: Identity: CC:8D:A2:ED:C5:64 (public)
[00:00:09.720,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:09.720,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
[00:00:09.721,000] <inf> smp_bt_sample: Advertising successfully started
[00:00:09.721,000] <inf> smp_sample: build time: Apr 28 2025 23:16:30
[00:00:09.722,000] <inf> smp_sample: boot_write_img_confirmed() appears to have been successful
[00:00:09.722,000] <err> smp_sample: boot_write_img_confirmed did not work -- this shouldn't happen
```

hobbes-400, 2025-05-07:
@rftafas moving this discussion here

almir-okato, 2025-05-24:
@hobbes-400 I've opened two draft PRs for two potential solutions, the branches needed by them for Zephyr and hal_espressif are indicated on each PR description:

https://github.com/mcu-tools/mcuboot/pull/2321

https://github.com/mcu-tools/mcuboot/pull/2320

Please note that there are still scenarios that need testing, and also some discussion may raise regarding which kind of solution fits MCUboot and Zephyr better. 

Could you please test your scenario with these?

closed 2025-10-01 by almir-okato