mcu-tools/mcuboot #2376: bootutil: sector size == padding size?

jorgenmk, 2025-07-04:
[https://github.com/mcu-tools/mcuboot/blob/main/boot/bootutil/src/bootutil_misc.c#L524](https://github.com/mcu-tools/mcuboot/blob/main/boot/bootutil/src/bootutil_misc.c#L524)

Here the padding size is assumed to always be the same as sector size, is this correct?

We are running into issues when application starts filling up the partition.

For example, try to build and run `sdk-nrf/applications/connectivity_bridge` for board `thingy91x/nrf5340/cpuapp`. This goes very wrong runtime.

Forcing padding_sz to 512 (default CONFIG_PM_PARTITION_SIZE_MCUBOOT_PAD) makes it work again.

@taltenbach

nordicjm, 2025-07-07:
268KiB was never right then. You have 272KiB slots, you need one extra sector in the primary slot for swapping so you are left with 268KiB, on top of this you have the flags and swap status, the size picked needs 64 sectors but let's assume 128 since NCS doesn't do auto configuration (check your MCUboot Kconfig for the real value) so 128 * 4 * 3 = 1536 bytes for swap status which rounds up to the sector size of 4KiB, 268 - 4 = 264KiB, which is 270336, so the new value is right and old value was wrong

maxd-nordic, 2025-07-07:
Alright, do we need to change what we pass to imgtool to make the build check effective or is this somewhere else?

maxd-nordic, 2025-07-09:
So this will just fail without indication now?
Also, how can it be that DFU succeeds on the previous bootloader versions if we should have needed this extra page?

nordicjm, 2025-07-09:
It can't, the swap sector is first used in the swap process then the other sector would be erased and written to with the swap status, if your had image data there then the image would be destroyed

nordicjm, 2025-07-14:
The check is not wrong, if it worked then you had different slot sizes or whatnot without realising it (or you're in upgrade only mode where there is no swap status), this is how it knows where it is when e.g. an unexpected reboot happens during update

nordicjm, 2025-12-30:
Looks like code has been reworked, assuming issue has been fixed in 2.3 release

closed 2025-12-30 by nordicjm